Jana-Marie/Otter-Iron-PRO #1: Why using an inductor?

rs-development, 2020-04-23:
Hey,

I am working on a similar project and was wondering why you are using an inductor for powering the iron? My guess, a fast PWM switching alone was not sufficient enough to maintain a decent USB-C Bus voltage? Maybe even triggering the power source?

Or are there some more reasons?

rs-development, 2020-04-23:
But don't get me wrong! Awesome work, ours is far more complex.... you did a good job keeping it simple!

jeanthom, 2020-04-23:
Hi!

I asked Jan Henrik about this:
> L1 is not determined yet, nor is determined if a inductor is needed. The topology with the inductor resembles a Buck-converter, such is may needed due to high pulse currents generated by the tip. However if I can ramp the driver frequency high enough, the inductor is not needed and can be replaced with a 0 Ohm resistor.

rs-development, 2020-04-23:
Ahh good to know, maybe I will try the supply our version with a USB C power supply soon to test this. Fell free to contact me if you would like to work together in some regard.