Jana-Marie/Otter-Iron-PRO #2: About L1 on v2.0 bottom PCB...

rsre, 2020-05-07:
Hi Jan,

A few friends and I were having a look at your design and we couldn't understand the purpose of L1 inductor on the Bottom PCB of v2.0.

Then we saw the picture of the assembled prototype and realised you had two 0 ohm resistors soldered there. Are you using them as fuses or for something else? Did you use the inductor symbol because it had the pad already designed?

Nice project over all, what are your thoughts regarding the design _doneness_?

Jan--Henrik, 2020-05-07:
Hey, I added L1 in case, if there is a energy storage needed (to reduce current spikes). However this is not the case and L1 can be bridged with a 0Ohm 2512 resistor.

I had an error on the "USB-PCB", which is now fixed but untested. Beside I think the design is more or less finished.

rsre, 2020-05-07:
Thanks for the quick answer. So, just to check, would it be OK to remove L1 from the design?

Jan--Henrik, 2020-05-07:
It is!

closed 2020-05-30 by Jan--Henrik