Jana-Marie/Otter-Iron-PRO #8: Add overheating protection/temperature amplifier saturation?

jeanthom, 2020-09-10:
Hi everyone,

Yesterday I ran into an issue with my Otter Iron PRO: the tip became glowing red hot! What happened is that I set its temperature to 400°C, and as the temperature readout was saturating at ~390°C the controller sent full power (65W) to the tip. What surprised me the most is that after letting the tip cool off, it still works (if I set the temp to < 390°C).

I'm using Jabe tips, not the JBC ones so that could explain the issue with the temp readout.

Has anyone here tried to run their iron at more than 390°C?

Jan--Henrik, 2020-09-13:
This is... ..new. I hope that it did not ruin your cartridge.
I will try to recreate the situation and find a fix. Is there anything else you noticed?

jeanthom, 2020-09-13:
To my biggest surprise, the tip is still functional! I was able to finish my VQFN bodge job, but I'm going to order a new one anyway just to be sure.

Yesterday I received some other kind of C245 tips. I still have the 390°C saturation issue, plus the temperature is super unstable (like ±20°C). I will try to investigate on my scope what's being sent to the ADC.

Jan--Henrik, 2020-09-13:
Please share the results! However I highly suspect my shitty control loop for being the cause. You might also want to try to tune it for better results