Jana-Marie/Otter-Iron-PRO #10: USB ! USB-PD Error Message

sleepiwill, 2020-10-10:
Hi Jan,

I'm trying to troubleshot the "USB ! USB-PD" message everything I power up my Otter Iron Pro using a Dell 90W USB C Power Adapter. 
Before I start pulling my hair out along with the components what do you recommend I check first to get the iron working?

Thanks,
Sleep

Jan--Henrik, 2020-10-24:
Hey,
despite the message, does the power supply put out 20V?

ElRoberto538, 2020-12-11:
I am getting the same issue using an 87w Macbook pro charger, it does support 20v 4.3A PD. Also tested on my work 60w USB-C screen, but I don't know what power profiles it supports. The Otter Iron only outputs 5v with this message showing. I checked all of the stusb connections and it looks good, can't find anything wrong physically. It is happening on both of the Otter Iron pros I've assembled.

@sleepiwill Did you solve this? I am about to update the firmware to display USB PD stats on the screen so I can debug this (I am an STM32 noob so not sure if there's an easier way).

ElRoberto538, 2020-12-13:
Ok, so I think I've solved this issue, [here](https://github.com/jeanthom/Otter-Iron-PRO-BOM/issues/7). I think we are both using the wrong USB interface chip!

swissfreek, 2020-12-19:
Glad I decided to troll the issues pages as I wait for parts for my build. The part number listed for LCSC is the "L" version, and they don't have a non-L version, but DigiKey has them, so I got a couple there.

closed 2021-04-19 by Jan--Henrik