Jana-Marie/Otter-Iron-PRO #11: Issues with Lenovo 4X20M26272 65w USB-C charger

jeanthom, 2020-10-18:
Hi,

I noticed a common issue among USB-C peripherals that are connected to this very specific model of charger: you can hear a faint whine noise, and there is a 1V ripple on VBUS (I'm probing C24 in AC mode in the picture below):

[image]

This is due to a too big capacitance on VBUS (USB spec specifies max 10µF). I remember fixing an FPGA board with a fat ass inductor but I'm not sure that this is a good solution.

This might explain the issues I'm facing in #8

Jan--Henrik, 2020-10-18:
Hey, I tried to replicate that and indeed it seems to be the same with my Lenovo 65W PD supply, but also my other supplies. This is a problem that needs to be fixed. Thank you for bringing this to my attention :)

hannesweisbach, 2020-10-20:
Did you capture that trace with or without the tip connected/on?

jeanthom, 2020-10-20:
With the tip on. When I disconnect the handle I don't have such oscillations.

hannesweisbach, 2020-10-20:
Ok.

If I see that right, the oszillations have a frequency of rougly 3 kHz. Does that track with the PWM frequency of the otter iron?