Jana-Marie/Otter-Iron-PRO #14: possibly killed by QC power supply?

martin2250, 2020-10-25:
So I was  experimenting with a power bank, which takes a long time to recognize the device and deliver power to the USB port. To test if the LCD powers on immediately when power is applied, I connected the Otter Iron to a "dumb" QC wall wart (12V max, USB-A to C cable), which should supply 5V regardless of whether or not a sink is attached. After a while, the board got hot around the STM/voltage regulator area. Now it doesn't power up anymore and always gets hot, even when connected to a PC. I can feel the STM32 itself warming up, which isn't a good sign.

Do you have any clue on what could have destroyed the STM32?

Jan--Henrik, 2021-06-13:
Hey, sorry for the late answer. Is this issue still active? It might be a defective LDO or cap

martin2250, 2021-06-13:
Hi, you did fix the board for me (thanks again!), but I don't think this issue was investigated further (I certainly won't test it again :D). From my end you can close this issue, but it might appear again...

closed 2021-06-14 by Jan--Henrik