Jana-Marie/Otter-Iron-PRO #16: Incorrect connection of the TC and heater

richard-sim, 2020-11-12:
Hey, I was looking over your schematic and it looks like you're powering the heater through the TC as they're connected in series in the cartridge with the 3 wires being either end and the common point between them. Going by the Bottom schematic, C245_JUNC is the common connection between the heater and the TC, so the heater should be powered by J3.1 (V+/heater) and J3.3 (C245_JUNC/common), and the TC sensed between J3.4 (TC) and J3.3 (C245_JUNC/common).

dreamcat4, 2020-11-12:
This **[eevBlog Forium thread](https://www.eevblog.com/forum/projects/yet-another-controller-for-jbc-t210t245/)** discussion goes into a lot of details about the C245 Tip. And is the best source of information I have found on the subject. It's probably worth the time to read the comments over there.

richard-sim, 2020-11-14:
There's an even better EEVblog forum post where someone has reverse engineered the entire circuit of a JBC CD-2BC: https://www.eevblog.com/forum/testgear/jbc-soldering-station-cd-2bc-complete-schematic-analysis/

If you look at that schematic, in the middle of sheet 2 is the TC. First thing it goes through is an analog switch that selects between various combinations of the 3 pins on the cartridge. Apparently for the T245 handle it selects TC+COM, and at startup and when a handle is plugged in it selects TC+TC (for zeroing?). I'm pretty certain that if they'd tried with a T210 handle they'd have found it selects LOAD+TC due to the different wiring of the cartridges.

Following that, the schematic also makes it clear that both C245 and C210 cartridges will have their heaters powered through the COM and LOAD pins only - never the TC/earth.

Here's a diagram I found in the Unisolder thread over at the Dangerous Prototypes forum. The T210/C210 connections do not match anything else that I've found online, however they fit the circuit above, and more importantly all my C210 cartridges corroberate it. As far as I can tell, all the "TC+heater are series only for C210" claims are wrong.

[image]

closed 2021-06-13 by Jan--Henrik