Jana-Marie/Otter-Iron-PRO #17: Debugging tips

ElRoberto538, 2020-12-06:
Hi, this is an awesome project (thanks!). I've managed to assemble mine after finally getting all of my parts, however I can't get it to boot into USB DFU mode, looking at the schematic it seems holding down SW2 while plugging it in should work, but I'm not seeing it listed using lsusb. I've checked a few pins and I can see that the STM32 is receiving 3.2v, and when holding down SW2 the boot pin also gets 3.2v.  One thing I noticed is that the current sense IC (U7) seems to be getting hot, I've checked most of the connections in that part of the circuit and nothing seems to be shorted. Any tips/recommended tutorials to debug this? I have some more parts so I can give it another go, but the soldering on the one I assembled looks good so I'd like to try and fix it.

ElRoberto538, 2020-12-06:
Well, I think I found the reason for the hot current sense IC, I accidentally used a 10M ohm not 10m ohm resistor... I don't think that would stop the STM32 booting though?

jeanthom, 2020-12-07:
I think your STM32 is toasted :confused: With R22 = 1Meg, you'll probably have the IN net at 24V-ish, which is directly fed into the STM32 ADC.

jeanthom, 2020-12-07:
Do you have an ST-Link debug adapter, like the ones you can find for dirt cheap on AliExpress?

ElRoberto538, 2020-12-07:
Ah yes, I didn't consider that... no I don't have one, I'll order one to try and flash it that way to confirm. Unfortunately shipping from china is fairly slow so I've started assembling a second one. I'll leave the sections separate (connect with wires first) this time so it's easier to debug! Do you see an issue with shorting R22 just to get it to boot and flash the firmware while I wait for new 10m resistors to arrive?

Thanks for your help!

jeanthom, 2020-12-07:
Where are you located? If you are in Europe, you should be able to order both the resistor and the STM32 chip from Arrow with free delivery.

Shorting R22 should be okay as long as you don't plug the tip in.

ElRoberto538, 2020-12-07:
I'm in New Zealand unfortunately, I can get free shipping but need to spend $65 minimum! I have enough parts left to complete a new one, so I'll see if I can flash a new one over USB by shorting R22 - thanks again for the help.

dreamcat4, 2020-12-07:
I looked on the arrow website today. However they do not seem to have such a good part availability as other suppliers? For example: which compatible STM32F072 they have? I do not see:

https://www.arrow.com/en/products/search?cat=&q=stm32f072&r=true

jeanthom, 2020-12-07:
It depends on the kind of part you're looking for. Sometimes they have huge stocks for an obscure part, no stock at all for a jellybean part, and the next day it's the opposite. They're on the expensive side, but with free shipping to EU and USA it remains a good deal.

@ElRoberto538 keep us in touch! :)

ElRoberto538, 2020-12-10:
Success! Kind-of, I made a second Otter Iron Pro, seems I'm getting the LCD reset bug though, 4 out of 5 times I plug it in it displays gibberish. Looks cool when it works though, just need to source a 10m Ohm resistor to test with an actual tip now.

How important is using a grounded USB-C power supply? Will I risk destroying sensitive components by soldering them with the iron tip un-grounded or is it a non-issue?

Thanks, @jeanthom, @Jan--Henrik!

[image]
[image]

jeanthom, 2020-12-10:
Congrats! I have the same OLED reset issue as you, but it's less frequent (I'd say 1 out of 5 times). #13 might have a fix.

My previous soldering iron was a DIY T12 kit from http://smdprutser.nl, which I used to solder my Otter Iron PRO. Most of the time it was not connected to ground, and so far it seems like I haven't destroyed things due to ESD. I'd say it's nice to have, especially if you're working with ESD sensitive devices, but not something critically important.

ElRoberto538, 2020-12-10:
Awesome, that's good to know, thanks for all your help.

closed 2020-12-10 by ElRoberto538