Jana-Marie/Otter-Iron-PRO #18: Free PCBs $2.50 shipping

ElRoberto538, 2020-12-08:
I ordered way too many PCBs (they only cost a few cents each!), so I can ship them to people for the cost of shipping (USD $2.50), which for some people will be a lot cheaper than shipping from JLCPCB or similar. I have 20 and should be able to fit multiple in a single envelope if anyone wants. They have black solder mask and lead free HASL. I can also chuck in some of the cheaper parts like transistors, resistors, capacitors etc.

If anyone is interested let me know and I'll put them up on Tindie or provide my email for details or something.

dreamcat4, 2020-12-08:
Hey @ElRoberto538, do you think you could ship to the UK? I could send you the shipping over PayPal. I would like to ask for 2 sets of PCB. If not, then no worries.

quentinmit, 2020-12-08:
I'd definitely be interested. My e-mail address is in my GitHub profile.

I suspect you'll have a lot of interest; you should probably do something like Tindie if it's easy.

dreamcat4, 2020-12-08:
Same here - my email address is in my GitHub profile.

ElRoberto538, 2020-12-08:
Great, I'll contact you both in the morning, it's getting late now. I can send globally via post (in paper envelope, will wrap in extra paper for durability) for NZD $3.50, OR roughly USD $2.50. Given the shipping method I can't make any promises on how well it will handle the trip, however as they're only PCBs I would be surprised if they didn't make it in one peice!

I don't have a PayPal account, but I have accounts in almost every country, or can set it up on Tindie as mentioned.

ElRoberto538, 2020-12-08:
Tindie listing is created - awaiting approval. https://www.tindie.com/products/22106/ @dreamcat4, @quentinmit please try to order via the listing - if it somehow sells out before you can order them I have reserved some for you.

dreamcat4, 2020-12-08:
Thanks! Please let us know once it goes live, by posting another comment here. Really appreciate it.

ElRoberto538, 2020-12-09:
Tindie listing is live; https://www.tindie.com/products/22106/

anuejn, 2020-12-09:
I also have some spare boards (in germany).
If you are interested I can ship them to you.

swissfreek, 2020-12-13:
> Tindie listing is live; https://www.tindie.com/products/22106/

Listing says out of stock. Are they already all spoken for?

I’m in the US, if anyone has extra boards/components, I’m interested in paying to take them off your hands.

ElRoberto538, 2020-12-13:
Hi @swissfreek, yes sorry they all sold out in about 12h. Maybe try contacting @anuejn?

swissfreek, 2020-12-13:
> Hi @swissfreek, yes sorry they all sold out in about 12h. Maybe try contacting @anuejn?

Not surprising, thanks anyway!

@anuejn anything left over? If not I guess I'll be the next person with extra boards to share ;)

swissfreek, 2020-12-15:
> yup I have some spare ones.
> any way to contact you?

Sweet!

swissfreek at gmail

TomHodson, 2021-03-03:
@anuejn Do you still have any left? I'm in Munich and would love to get some! My email is thomas c hodson at gmail and it's also on my github.

danikdd, 2022-02-09:
> yup I have some spare ones. any way to contact you?

Hi there, any spare pls? I have adress in Germany or Czech Republic. many thx for info. Nice day. Daniel

m-kozlowski, 2022-02-09:
PCBs? no problem. but might have trouble getting all the other stuff - my orders for USB-PD controller at tme and farnell are being rescheduled every few weeks for over a year(!)

ElRoberto538, 2022-02-09:
I have 4 STUSB4500QTR USB-PD left over, and I think 6 STM32's. I could send them in an envelope (about ~$1) if someone wants, no guarantees they'll arrive though as I got told off last time by the post service as apparently envelopes are for letters only...

sorki, 2022-07-19:
> I have 4 STUSB4500QTR USB-PD left over, and I think 6 STM32's. I could send them in an envelope (about ~$2) if someone wants, no guarantees they'll arrive though as I got told off last time by the post service as apparently envelopes are for letters only...
> 
> Edit: Also looks like I still have some of the other parts like op-amps, mosfets, esd suppression diodes etc

Hi, are these still available? I'm considering a small run of 5 boards but STUSB and STM32F1s are out of stock (available next year..).

ElRoberto538, 2022-07-21:
> Hi, are these still available? I'm considering a small run of 5 boards but STUSB and STM32F1s are out of stock (available next year..).

Yes I still have them, just checked and I also have 4x sets of PCBs here too. I have Covid at the moment so am in quarantine, but I could post them in a week or so if you send me a way to contact you for address details.

closed 2022-07-28 by ElRoberto538