Jana-Marie/Otter-Iron-PRO #19: A question about OLED display

snowman-andy-y, 2020-12-13:
Otter iron Pro is a great idea! I tried to make the PCB and verify it. At present, I only used Huawei mobile phone charger for testing. Unfortunately, it can't display normally: only random points are displayed on the display. I don't know my OLED monitor is compatible or not, I am asking the supplier to provide the manual. I noticed that Martin mentioned this situation in his question. But, I can't display it normally so far.

I also have a question about firmware. If I want to modify firmware, what kind of programming environment do I need? Maybe I have to modify the firmware code to display properly.
[image]

ElRoberto538, 2020-12-13:
Hi @snowman-andy-y, I have the same issue, try turning it on and off multiple times - mine only works about every 1 in 5-10 times I turn it on. It is a known issue - check [here](https://github.com/Jan--Henrik/Otter-Iron) and scroll to known issues which shows soldering a 100nf capacitor from the LCD reset to ground helps. I haven't done this myself yet, so if you do, please let me know how it goes.

snowman-andy-y, 2020-12-14:
> Hi @snowman-andy-y, I have the same issue, try turning it on and off multiple times - mine only works about every 1 in 5-10 times I turn it on. It is a known issue - check [...]

Thank you very much for your advice! It works! I added 100nF between OLED reset and GND, and the display works normally!

closed 2020-12-24 by snowman-andy-y