Jana-Marie/Otter-Iron-PRO #20: Firmware difference to Otter-Iron (non-pro)

NeoFromMatrix, 2020-12-18:
I wanted to ask if the Otter-Iron-PRO hardware is compatible with the Otter-Iron non-PRO firmware (https://github.com/Jan--Henrik/Otter-Iron).
It seems that the non-PRO has developed a bit further while this repo stalled. 

If it is compatible, or one firmware for both hardware versions is desired, do you want to add a note in this repo pointing to the other repo? 

Also, thanks a lot for your work, I highly appreciate it! :)

ElRoberto538, 2021-01-17:
From looking at the code, the firmware seems compatible, however it runs at lower power... the updates are to support lower power profiles. I copied them into the pro firmware for testing as I only had a 28w adapter. With the lower power it works fine but takes a long time (30+ seconds) to heat up. Having said that I did borrow a 100w apple adapter and changed the firmware to use the full 100w - it works great! 

Unless you need lower power I'd stick with the pro firmware.

Jan--Henrik, 2021-04-19:
Hey! No they are NOT compatible, the PROs PWM and PID loop run at a much higher frequency! Please use the PRO firmware for the PRO.

Both firmwares need bugfixes and the PRO firmware needs to be updated to the non-pro firmware version. I will add that to my todo list :)

closed 2021-06-13 by Jan--Henrik