Jana-Marie/Otter-Iron-PRO #22: Firmware programming?

bb-centrum, 2021-04-05:
Hi, I'm new to programming STM32 micro's and could do with some help. Do you have any instructions to follow now my device is all built?

Thanks

Jan--Henrik, 2021-04-05:
Hey!
Please follow the programming instructions for the Otter-Iron, they are the same.
https://github.com/Jan--Henrik/Otter-Iron#flashing

bb-centrum, 2021-04-05:
Perfect, thank you! All finished and working. Now just to buy the handle!

closed 2021-04-05 by bb-centrum

swissfreek, 2021-04-19:
Sorry, maybe I'm being dense. I saw the instructions on the Otter Iron page, but I don't see where the DFU pins are on the Otter Iron Pro. I can't hold the buttons because it hasn't been flashed yet, right? Thoughts?

jeanthom, 2021-04-19:
Hi, there is no pin to short. You just have to press one of the buttons (don't remember which one) while plugging the Otter-Iron to your computer.

swissfreek, 2021-04-19:
> Hi, there is no pin to short. You just have to press one of the buttons (don't remember which one) while plugging the Otter-Iron to your computer.

Ah yes, of course, the schematic shows SW2 is connected to the boot pin. Thanks! Found out my issue wasn't the board, but the fact that the Mac's USB-C ports are super finicky about only having "real" USB-C devices connected to them. When I plugged the iron in through a USB-A adapter (so, C->A->C...) it worked and I was able to flash.