Jana-Marie/Otter-Iron-PRO #24: Troubleshooting Help?

swissfreek, 2021-04-23:
I built my Otter Iron Pro. It seems that everything is in order. It accepted the firmware when I flashed it, so the processor seems to be OK and wired correctly. When I plug it into a PD power supply, I can probe the pins of the capacitor and I get 20.0V, so it seems the PD circuitry seems to be working. But I get nothing on the OLED. Not gibberish, it just doesn't power up. I initially bought the OLEDs from LCSC because why not, they're so cheap. So then I assumed that I got the bogus ones as noted on the Otter Iron page, and bought another pair of Amazon. I swapped them out but get the same result. I assume that they are the same crap components as from LCSC. Not sure where else to get these OLEDs. Curious if anyone has any thoughts. I have a JBC handle, but still waiting on the tips, so I'm not sure if pushing buttons will make the iron heat up. I welcome any thoughts on what I could try to troubleshoot the iron. Thanks!

swissfreek, 2021-04-23:
Found one at DigiKey, so hopefully a reputable component (and definitely not cheap, relatively speaking). https://www.digikey.com/en/products/detail/display-visions/EA-W096016-XALW/8021564

Will try this as well when it arrives... 🤷🏼‍♂️

jeanthom, 2021-04-23:
Out of the two OLED displays I bought at LCSC, one of them was defective...

swissfreek, 2021-04-25:
OK, I guess I will see if I have better luck with the ones I got from DigiKey. This would be 0 out of 4 for me...

Jan--Henrik, 2021-06-13:
@swissfreek any news? :)

swissfreek, 2021-06-13:
No news. Neither iron works properly. I was able to flash one. The other I’m pretty sure I had a bad solder joint in the USB-C connector that was causing a short and now it won’t respond, so I assume I fried the processor. Who knows when I’ll be able to get a new one.

the one I can flash appears to take firmware successfully, but I get nothing on the display when I power it up. Honestly just haven’t had the motivation to dig into it more after all the time I spent building both of them and having them both not work.