Jana-Marie/Otter-Iron-PRO #25: Iron heater never powers on.

bb-centrum, 2021-04-24:
Hi!

I have the iron working in the sense it powers on, displays the PD profile info, and then says that it is Active but the heater never comes on. The current temp seems to be working fine. I probed pins 4 and 1 on the connector between the bottom board and the C245 connector board and there was no voltage. Probing between GND and pin 5 on the DMP3017SFG gave zero volts, pin 4 = 0.3v and pin 1 = 0.3v

So something is off somewhere, I'm just not sure where to look next. Any ideas?

m-kozlowski, 2021-04-24:
Sure it says it's active, or "!Active"?
When iron is really active (or at least it thinks it is) there is a graphic Bar, not a text.

Which PD profile is negotiated at boot?

bb-centrum, 2021-04-24:
> Sure it says it's active, or "!Active"?
> When iron is really active (or at least it thinks it is) there is a graphic Bar, not a text.
> 
> Which PD profile is negotiated at boot?

I have two USB C power supplies.

An Anker PowerCore. This negotiates PD1. No graphic bar, it just says !Active (not active)?

I also have a 30W Macbook PSU. This says PD Adjust 4 pm boot. If the iron is not connected then I get a loading bar. Connect the iron and I get !Active again. So I think the Anker one is just not powerful enough, but there looks like there is another issue with still!?

m-kozlowski, 2021-04-24:
Is there any chance that you used 10Mohm as R22 instead of 10mOhm?

bb-centrum, 2021-04-24:
Ooh good shout. I'll check that tomorrow morning

bb-centrum, 2021-04-25:
Yup 10M ohm! I'll get that replaced and then hopefully I'll have a working iron!

jeanthom, 2021-04-25:
Some else had the same issue as you (10Mohms instead of 10mOhms), and they had to change the micro-controller: https://github.com/Jan--Henrik/Otter-Iron-PRO/issues/17

m-kozlowski, 2021-04-25:
Yeah, si8540 output is capped at 8V, but it's still enough to fry stm32

bb-centrum, 2021-04-25:
Well I managed to program the stm32 and it boots all ok, so hopefully mine is fine somehow. I’ll let you know when I get the replacement resistor tomorrow

closed 2021-06-13 by Jan--Henrik