Jana-Marie/Otter-Iron-PRO #28: Question: STUSB4500 out of stock - replace with a STUSB4761?

Knarf2807, 2022-11-15:
The STUSB4500 is in all shops I looked for out of stock. I wonder, if I replace ist with a STUSB4761? Another Footprint and another Pinout so I have to change the schematic and the PCB. 

I think it will work. What do you think? And why do you connect the VSYS to +3V3? Is it necessery for the OtterIron Pro, because the STUSB4761 has no VSYS like the STUSB4500? I found no reason why it is necessery.

Thanks a lot for your help.

Frank

Jana-Marie, 2022-11-15:
Hey, no the STUSB4500 cant be replaced with STUSB4761. The latter is a source controller, not a sink controller.

closed 2022-11-15 by Jana-Marie

Knarf2807, 2022-11-18:
I will try a STUSB4500BJRCT (BGA). Because it's my first time "soldering" BGA, I'm not shure, that it works after my soldering. That's why I will build in a "bypass" and a normal round DC-Plug. I have to think over my idea, but at the moment it's the best I have (can't wait o.O )